hdl/bazel_rules_hdl #10: Emulation: Get Renode into Bazel rules

mithro, 2020-12-05:


mithro, 2020-12-05:
@mgielda @pgielda @kgugala

mithro, 2020-12-05:
@PiotrZierhoffer -- I think Renode portable is the way to go here?

mithro, 2020-12-05:
As Renode is not strictly HDL related, maybe there should be a seperate https://github.com/renode/bazel_rules_renode? It would be good if the bazel rules supported Verilator co-simulation so probably would want to at least reference this.

umarcor, 2020-12-06:
> As Renode is not strictly HDL related

I believe github.com/hdl is for "whatever you want to achieve with the HDL sources you have/write". Hence, Renode fits in the context, without the need for a separate repo. However, that repo might exist and be added as a submodule here.

umarcor, 2020-12-06:
Ref: [hdl.github.io/awesome/items/renode](https://hdl.github.io/awesome/items/renode/)

mithro, 2022-03-10:
FYI - @kgugala / @mgielda - I just rediscovered this. Should `bazel_rules_hdl` be including `bazel_rules_renode` in some way and how would this work with co-simulation?