hdl/bazel_rules_hdl #17: Language: Create good support for VHDL

mithro, 2020-12-05:


mithro, 2020-12-05:
@umarcor -- This is your cup of tea, right?

umarcor, 2020-12-06:
It is. It's as "simple" as providing GHDL (for simulation) plus ghdl-yosys-plugin for synthesis and formal verification. Hence, this issue depends on #2 (Yosys) and #9 (Symbiyosys). For mixed-language simulation, it can be combined with Verilator (#5): https://github.com/ghdl/ghdl/issues/1512#issuecomment-731712474.

solsjo, 2022-06-19:
Hi!

Stumbled on this issue and wondered if it could be of help.

I wrote Bazel rules for vunit and ghdl, albeit only for simulation, not for synthesis. It builds ghdl from source but is dependent on llvm and gnat.

[rules_ghdl](https://github.com/hdl/bazel_rules_hdl/issues/93)

Though I don't have a gut feeling for the work needed to integrate with ghdl yosys.

Just thought it might be of interest.

mithro, 2022-06-20:
@solsjo - It would be great to add this to this repo even if only for simulation.

filmil, 2025-10-05:
Here's a modules-ready hermetic ghdl set of bazel rules: https://github.com/filmil/bazel_rules_ghdl, in case that is useful to you.