hdl/bazel_rules_hdl #19: ASIC: Get Magic into bazel

mithro, 2020-12-05:
Needed for things like DRC checks and GDS generation.

umarcor, 2020-12-06:
Ref: [hdl.github.io/awesome/items/magic](https://hdl.github.io/awesome/items/magic/)

mithro, 2020-12-14:
@per-gron apparently has rules for Magic.

per-gron, 2020-12-14:
Yep, I can do this one.

per-gron, 2020-12-16:
As of 69c11fec930f1f1b5b6c385a07cb3a3d69d765d0 Magic is available as `@com_opencircuitdesign_magic//:magic`. This is however limited to the non-Tcl version of Magic because I didn't want to pull in the Tk and GUI dependencies. I don't know if this is enough for what you want to use it for.

mithro, 2020-12-16:
I think we really do want the tcl support so we can do automated DRC checking and automatic format conversion. What confuses me is why the TCL support requires a GUI.

per-gron, 2020-12-17:
AFAIU the non-Tcl version of Magic still supports these features, it just doesn't support Tcl scripting.

I don't think there's a fundamental reason for Tcl support requiring a GUI, it's just that Magic is currently written to be compilable in two modes: without Tcl as a command line tool, or with Tcl/Tk with GUI support. There's enough interdependencies between Tcl code and GUI code that it's not practical to build Magic with Tcl without GUI without doing Magic code changes (it would require more patches than I think makes sense to maintain in this repo).