hdl/bazel_rules_hdl #3: FPGA: Get Yosys+nextpnr+XXX flow into bazel

mithro, 2020-12-05:
This depends on https://github.com/hdl/bazel_rules_hdl/issues/1

umarcor, 2020-12-06:
Refs:

- [hdl.github.io/awesome/items/yosys](https://hdl.github.io/awesome/items/yosys/)
- [hdl.github.io/awesome/items/nextpnr](https://hdl.github.io/awesome/items/nextpnr/)
- [hdl.github.io/awesome/items/icestorm](https://hdl.github.io/awesome/items/icestorm/)
- [hdl.github.io/awesome/items/prjtrellis](https://hdl.github.io/awesome/items/prjtrellis/)
- [hdl.github.io/awesome/items/prjoxide](https://hdl.github.io/awesome/items/prjoxide/)
- [hdl.github.io/awesome/items/mistral](https://hdl.github.io/awesome/items/mistral/)

per-gron, 2020-12-14:
There are rules for

* nextpnr https://github.com/google/xls/tree/main/dependency_support/nextpnr
* icestorm https://github.com/google/xls/tree/main/dependency_support/at_clifford_icestorm
* prjtrellis https://github.com/google/xls/tree/main/dependency_support/prjtrellis https://github.com/google/xls/tree/main/dependency_support/prjtrellis_db

per-gron, 2021-01-21:
Here is a C++ program in XLS that calls Yosys and Nextpnr to synthesise verilog code for Lattice FPGAs: https://github.com/google/xls/blob/main/xls/synthesis/yosys/yosys_server_main.cc