hdl/bazel_rules_hdl #5: Simulation: Get Verilator into Bazel for simulation

mithro, 2020-12-05:
I think there is some stuff in https://github.com/google/xls/tree/main/xls/build_rules ?

umarcor, 2020-12-06:
Ref: [hdl.github.io/awesome/items/verilator](https://hdl.github.io/awesome/items/verilator/)

kkiningh, 2020-12-06:
I have some rules I would be happy to contribute. See https://github.com/kkiningh/rules_verilator

umarcor, 2020-12-08:
@kkiningh that's nice!

@mithro, any guidelines about the structure you want to use in this repo?

QuantamHD, 2020-12-08:
I think the first thing we need to get going is a VerilogInfo provider. The benefit of that is all rules can target it for physical processing. That's how I've been formatting it within Google.

umarcor, 2020-12-08:
@QuantamHD, was that comment meant to be posted here? It seems to be related to Verilog (#15) and/or SystemVerilog (#16) rather than Verilator...

QuantamHD, 2020-12-08:
For example the verible rule set in question defines a VerilogInfo as https://github.com/kkiningh/rules_verilator/blob/master/verilator/defs.bzl#L12

I think what I'd like to see is an authoritative provider other rules can target or consume; Verilog or not within this repo.  HDLInfo or something related. We can provide the transitive src resolution etc. so that other rules don't need to deal with it as much. I just want a solid foundation so that we can add useful tools such that we can get solid compatibility with all the tools.

kkiningh, 2020-12-08:
@agoessling has been doing some work in this direction in his [rules_verilog](https://github.com/agoessling/rules_verilog) repo. The corresponding provider is VerilogModuleInfo and it should be supported in rules_verilator already.

QuantamHD, 2020-12-08:
Awesome! I'm glad there's something out there we can build around. Thanks for the tip @kkiningh.

agoessling, 2020-12-08:
@QuantamHD `VerilogModuleInfo` is currently used in:
- [rules_verilator](https://github.com/kkiningh/rules_verilator)
- [rules_symbiyosys](https://github.com/agoessling/rules_symbiyosys)
- [rules_vivado](https://github.com/agoessling/rules_vivado)

Currently it is quite basic and only provides sources and the top module name.  The idea being this can be shared between many tools and users only need to define their module structure once.

mithro, 2020-12-08:
@agoessling - Would love to have your help in building this out.

agoessling, 2020-12-09:
What is "this"?

QuantamHD, 2020-12-09:
"this" is a project is supposed to be a one stop shop for doing hardware dev with bazel.

So that includes

Having rules that take can RTL and be chained to emit GDSII (Sky130/FOSS-PDKs), but that can also be used to target FPGAs and soft CPUs. We want to take advantage of [RBE](https://blog.bazel.build/2018/10/05/remote-build-execution.html) to accelerate verification with massively parallel clusters. Using the open source community's tools like Yosys, OpenROAD, Verilator, Surlog etc. 

My vision of this is that we enable `bazel test //mychip/...` and `bazel build //mychip:gds` and in 10 minutes you have verified and emitted a fabricatable design even for the most complex designs.

per-gron, 2020-12-16:
@QuantamHD I think it could be inspiring and generally useful if you added basically what you wrote in that last comment to the top level README.md :-)

PR #115 (2022-10-14): Port kkiningh/rules_verilator to rules_hdl.
* Add default prefix to flex.bzl
* Add example/test in //verilator/tests:adder_test

Fixes #5
closed 2022-10-14 by QuantamHD