hdl/bazel_rules_hdl #26: Setup GitHub Actions CI

mithro, 2020-12-16:
This repository is quickly going to need CI.

hzeller, 2020-12-17:
Per volunteered in #24

per-gron, 2020-12-17:
@mithro Do you want to use GitHub Actions specifically? I suggested using Google Cloud Build because 1) I figured it might be easier for us to get quota there, 2) I think if we're going to run Bazel Remote Execution and caching stuff it seems likely that running the execution nodes in the same cloud as the CI is going to help performance.

per-gron, 2020-12-17:
GitHub Actions uses VMs with 2 CPUs and 14 GB of disk space, both of which seem kind of low. Cloud Build allows choosing beefy worker VMs.

mithro, 2020-12-17:
@per-gron I would say that GitHub Action is the "default choice unless a strong reason otherwise".

per-gron, 2020-12-17:
@mithro I see, that makes sense. But I like fast CI machines... Let me present one argument for Cloud Build: After completely cleaning up my Bazel cache and doing `bazel build @tk_tcl//... @com_github_yosyshq_nextpnr//... @com_github_yosyshq_prjtrellis//... @org_tuxfamily_eigen//... @com_google_skywater_pdk//... @com_icarus_iverilog//... @com_opencircuitdesign_magic//... @com_opencircuitdesign_netgen//... @net_sourceforge_ngspice//...` the Bazel cache uses 26GB of disk space. Compare this with the 14GB of disk space of GitHub Actions VMs 😉 And there are many more dependencies that haven't yet been added.

In general, hardware synthesis tools don't really seem to agree with weak computers.

Can I haz beefy CI pleeeease??

[image]

mithro, 2020-12-20:
Yes, I'll look at setting this Cloud Build then...

mithro, 2020-12-20:
I believe this is now set up but I'm unclear how to test it?

[image]

mithro, 2020-12-20:
[image]

per-gron, 2020-12-21:
Cool! I'll see if I can make it do something useful.

per-gron, 2020-12-21:
@mithro In #30 I got Cloud Build to do a successful "build", as in running `echo This is a test`. However, I don't have access to build logs. I presume that we want everyone to have read access to these build logs. Without access to the build logs I don't think I'll be able to create a non-trivial build configuration.

I could fix this if you grant me full "Project > Owner" access to the project. Alternatively you could grant me or everyone read access to the logs. This requires two roles: [roles/cloudbuild.builds.viewer](https://cloud.google.com/cloud-build/docs/iam-roles-permissions) to get access to the Cloud Build web UI at all, and then it's also necessary to [grant access to the log files](https://cloud.google.com/cloud-build/docs/view-build-results).

per-gron, 2020-12-23:
@mithro It seems that Cloud Build is not compatible with open source projects because it offers no way to make build logs public. It seems that our options are:

1. Use Cloud Build anyway and accept that only Googlers/core contributors will be able to view build logs. This is obviously not great but it might be an acceptable temporary solution. For security reasons we're likely going to require that a trusted person trigger the build after initial code review anyway so this is not quite as terrible as it sounds at first, we can manually share URLs to build logs when necessary.
2. Use a separate CI system (perhaps GitHub Actions) to either trigger Cloud Build or otherwise extract Cloud Build logs. There's no easy way of doing this.
3. Use some other CI system. As I said above, I don't think GitHub Actions would work because of the limited disk space on their machines (and they offer no way to upgrade), so we'd have to use something else. I think none of the free options work, the free VMs are too small (but I haven't checked all options).

umarcor, 2020-12-23:
> Use a separate CI system (perhaps GitHub Actions) to either trigger Cloud Build or otherwise extract Cloud Build logs. There's no easy way of doing this.

What's the technical limitation for making GHA trigger Cloud Build and wait until completion through pooling? We can have secrets in GHA for interacting with any API. The limitation would be that PRs would not have access to those secrets.

per-gron, 2020-12-23:
@umarcor This is possible to do, but it seems like a lot more work than I expected to be able to see build logs... I don't expect to need to use a CI system to get another CI system to make the logs accessible.

It seems like it might be possible to use [Cloud Build Notifiers](https://github.com/googlecloudplatform/cloud-build-notifiers) for this. Such a piece of code can be programmed to give the build logs to GitHub so that they are accessible directly from the PR. Look at this: https://github.com/wlynch/test/pull/1/checks I am talking to wlynch about how to set this up. Will report back in early January.

PR #30 (2021-01-14): Use Cloud Build for continuous integration
Fixes #26.
closed 2021-01-14 by QuantamHD