hdl/bazel_rules_hdl #73: Compiler -D argument escape issues with newer Bazel (gcc: error: .: No such file or directory)

jesec, 2021-12-17:
Bazel 4.2.2, GCC 9.3.0

at_clifford_yosys:

```
gcc: error: ?: No such file or directory
gcc: error: std::getenv("YOSYS_DATDIR"): No such file or directory
gcc: error: :: No such file or directory
gcc: error: ".": No such file or directory
```

arguments: `'-DYOSYS_SRC="./"' '-DYOSYS_DATDIR=std::getenv("YOSYS_DATDIR") ? std::getenv("YOSYS_DATDIR") : "."'`

source: https://github.com/hdl/bazel_rules_hdl/blob/9192cbcd04fee629492f906fb443c4bc760e1745/dependency_support/at_clifford_yosys/bundled.BUILD.bazel#L129-L130

QuantamHD, 2021-12-18:
@jesec Could you try  

`"-DYOSYS_DATDIR=\"std::getenv(\"YOSYS_DATDIR\") ? std::getenv(\"YOSYS_DATDIR\") : \".\"\"", `

QuantamHD, 2021-12-18:
If you could try a few different escape options and let me know what works that would really help us out. I don't have access to a computer with bazel at the moment.

QuantamHD, 2021-12-20:
I can't reproduce this on bazel 4.2.2 at 184e7865e2f8b182a13f213c214aec1bc53aeee7

jesec, 2021-12-20:
Interesting. Let me try again.

jesec, 2021-12-20:
It is still there.

> @jesec Could you try
> 
> `"-DYOSYS_DATDIR=\"std::getenv(\"YOSYS_DATDIR\") ? std::getenv(\"YOSYS_DATDIR\") : \".\"\"", `

and this does not work.

I can confirm with clang as well:

```
'-DYOSYS_DATDIR=std::getenv(YOSYS_DATDIR) ? std::getenv(YOSYS_DATDIR) : .'
```

```
external/at_clifford_yosys/kernel/yosys.cc:867:20: error: expected expression
        proc_share_path = YOSYS_DATDIR "/";
                          ^
<command line>:18:78: note: expanded from here
#define YOSYS_DATDIR std::getenv(YOSYS_DATDIR) ? std::getenv(YOSYS_DATDIR) : .
                                                                             ^
external/at_clifford_yosys/kernel/yosys.cc:867:20: error: use of undeclared identifier 'YOSYS_DATDIR'
<command line>:18:34: note: expanded from here
#define YOSYS_DATDIR std::getenv(YOSYS_DATDIR) ? std::getenv(YOSYS_DATDIR) : .
                                 ^
external/at_clifford_yosys/kernel/yosys.cc:867:20: error: use of undeclared identifier 'YOSYS_DATDIR'
<command line>:18:62: note: expanded from here
#define YOSYS_DATDIR std::getenv(YOSYS_DATDIR) ? std::getenv(YOSYS_DATDIR) : .
                                                             ^
3 errors generated.
```

Somehow the escape does not work as intended.

jesec, 2021-12-21:
Yeah. It is probably a Bazel regression. It could be in rules_cc. 

```
r'YOSYS_DATDIR=' + "\"" + r'std::getenv(\"YOSYS_DATDIR\")' + r' ? ' + r'std::getenv(\"YOSYS_DATDIR\")' + r' : ' + r'\".\"' + "\""
```

yields

```
'-DYOSYS_DATDIR=std::getenv("YOSYS_DATDIR") ? std::getenv("YOSYS_DATDIR") : "."' '-DYOSYS_SRC="."'
```

It is better than the current one, but still incorrect.

```
'\\"' or  r'\"'
```

makes Bazel complain that `definition contains too many tokens (found 5, expecting exactly one)`.

It is probably a bad idea to inject a complicated statement with `-D` anyways. IMO the best option is to use a patch.

QuantamHD, 2021-12-21:
I still can't reproduce it with GCC or clang, but I think that this should resolve it. https://github.com/hdl/bazel_rules_hdl/pull/74/files

jesec, 2021-12-21:
> I still can't reproduce it with GCC or clang, but I think that this should resolve it. #74

With #74, the result is better but still incorrect: 

```
'-DYOSYS_DATDIR=std::getenv("YOSYS_DATDIR") ? std::getenv("YOSYS_DATDIR") : "."'
```

QuantamHD, 2021-12-21:
This looks correct to me though. Does this still not compile for you?

jesec, 2021-12-21:
The same error. 

It should be `-DYOSYS_DATDIR="std::getenv("YOSYS_DATDIR") ? std::getenv("YOSYS_DATDIR") : ".""` (note the added ").

jesec, 2021-12-21:
I have already tried to workaround the issue (see https://github.com/hdl/bazel_rules_hdl/issues/73#issuecomment-998375146), which yields the same result as #74.

However, that's not enough and I haven't be able to figure out a way yet.

PR #75 (2021-12-21): at_clifford_yosys: replace ternary "-D" with patch, ncurses: cp with "-f" to deal with unsandboxed cases 
at_clifford_yosys: replace ternary "-D" with patch (fix #73):

```
There seems to be a regression (or behavior change) with Bazel 4.2.2 in string escaping or copts sanitization.
```

ncurses: cp with "-f" to deal with unsandboxed cases:

```
Fixes: "cp: cannot create regular file './make_hash': File exists"

This issue only happens when Bazel runs with unsandboxed strategies
like "local" or "standalone".

On some platforms (e.g. macOS), debug symbols of compiled binaries
are stored in sepa [... cut]
closed 2021-12-21 by QuantamHD