hdl/bazel_rules_hdl #93: Ghdl and VUnit rules

solsjo, 2022-02-18:
Hi!
I’ve written  bazel rules for ghdl and vunit
it's version 0.1, so there is some substantial work left

But it might be of interest for you.

https://github.com/solsjo/rules_ghdl
https://github.com/solsjo/rules_vunit
https://github.com/solsjo/ghdl-example

There are some prerequisites, such as that it only runs on linux, and gnat and llvm needs to be installed,
but other than that it build ghdl from source.

would it be okay to add to this repository?

QuantamHD, 2022-02-18:
Absolutely! Gnat would need to be built from source,  but we can take that one step at a time.

solsjo, 2022-02-18:
Cool!
What about llvm?
Do you build that from source or is it provided by Bazel as a cc toolchain?

QuantamHD, 2022-02-18:
LLVM is currently provided as a toolchain. If we need to I'm happy to get LLVM building from source in the repo.

sam-bristow, 2023-03-17:
@solsjo it looks like your `rules_vunit` repo has disappeared. Is it something you can bring back easily?

solsjo, 2023-05-07:
Hi @sam-bristow !

Sorry for the late reply, yes I should be able to bring it back up again. Though it make take some time.

I'll try to keep you posted.

Kind regards,
Oskar Solsjö

filmil, 2025-10-05:
Hi folks. 

Here's a repo that defines a hermetic bazel module for ghdl: https://github.com/filmil/bazel_rules_ghdl