hdl/bazel_rules_hdl #111: m4 compile issue

renau, 2022-09-07:
When using the latest libc, the m4 packages does not compile:
```
ERROR: ..../1c203eacaa58ab62be36bb8031d3e676/external/m4_v1.4.18/gnulib/BUILD.bazel:192:11: Compiling gnulib/lib/c-stack.c [for host] failed: (Exit 1): clang failed: error executing command external/llvm_toolchain/bin/clang -U_FORTIFY_SOURCE -fstack-protector -fno-omit-frame-pointer -fcolor-diagnostics -Wall -Wthread-safety -Wself-assign -g0 -O2 '-D_FORTIFY_SOURCE=1' -DNDEBUG ... (remaining 26 arguments skipped)

Use --sandbox_debug to see verbose messages from the sandbox and retain the sandbox build root for debugging
external/m4_v1.4.18/gnulib/lib/c-stack.c:55:26: error: function-like macro 'sysconf' is not defined
#elif HAVE_LIBSIGSEGV && SIGSTKSZ < 16384
                         ^
/usr/include/x86_64-linux-gnu/bits/sigstksz.h:28:19: note: expanded from macro 'SIGSTKSZ'
# define SIGSTKSZ sysconf (_SC_SIGSTKSZ)
                  ^
external/m4_v1.4.18/gnulib/lib/c-stack.c:139:8: error: fields must have a constant size: 'variable length array in structure' extension will never be supported
  char buffer[SIGSTKSZ];
       ^
2 errors generated.
```

The https://github.com/jmillikin/rules_m4 package has a patch for this, but the custom org_gnu_m4 package inside bazel_hdl_rules does not.

https://github.com/jmillikin/rules_m4/blob/7bff5f2b25055c4403412793857ab6c2d5cd21fd/m4/internal/gnulib/gnulib.bzl#L143

Maybe switching to jmillikin/rules_m4?

renau, 2022-09-07:
In the MASC group branch, we did a more ugly patch:

https://github.com/hdl/bazel_rules_hdl/commit/32245a569eaa0dd863770b4f483776f20eee6696

hzeller, 2022-09-07:
Interestingly we already have [a patch](https://github.com/hdl/bazel_rules_hdl/blob/3497bab04e37514dcdeb47e3090d57d78e2aa01a/dependency_support/org_gnu_gnulib/stop_using_sigstksz.patch) for gnulib to not to use the `SIGSTKSZ` macro.

And the `m4` compile should [actually use that library](https://github.com/hdl/bazel_rules_hdl/blob/3497bab04e37514dcdeb47e3090d57d78e2aa01a/dependency_support/org_gnu_m4/bundled.BUILD.bazel#L35) instead of compiling gnulib inside m4.

Investigating...

In the meantime, question for you @renau :
 * Are you using  `bazel_rules_hdl` from head ?
 * What was the bazel build target were you building ?

renau, 2022-09-07:
It is a separate repo (you can clone and check) that uses bazel_rules_hdl and also verilator bazel). It runs synthesis and run gate level simulation.

https://github.com/renau/bazel_rules_hdl_test/

If you want, check the counter target:
```
bazel build //counter:top_gls
```

hzeller, 2022-09-07:
In your WORKSPACE file, you're also including rules_m4: https://github.com/renau/bazel_rules_hdl_test/blob/main/WORKSPACE#L101-L109

Could it be that your m4 compile issue comes from that location ? The location looks like it is not the bazel_rules_hdl location.

What happens if you compile `m4` from bazel_rules_hdl ?
```
bazel build @org_gnu_m4//:m4
```

renau, 2022-09-07:
My mistake, the `bazel build @org_gnu_m4//:m4` works fine. Looks like the bazel verilator is the one pulling an old version.

I'll create there the pull request.

closed 2022-09-07 by renau