hdl/bazel_rules_hdl #123: Issues with README.md instructions

tcal-x, 2022-12-13:
I tried running the example in the README.md.   I have bazel 5.3.2 installed.

I created `WORKSPACE` with the exact text in the README, except for:
```
rules_hdl_git_hash = "ebaf7482c035208f485f463c62fd3c2f969a9b5c"
rules_hdl_git_sha256 = "3743f1ed6739abaaa68e1e907adffb13c285fd70390d950c3989729439d952c5"
```

Perhaps the provided code was never meant to be the *entire* WORKSPACE, but for this example, I assumed it was.

I encountered:
```
ERROR: /usr/local/google/home/tcal/bazel/WORKSPACE:11:1: name 'maybe' is not defined
ERROR: /usr/local/google/home/tcal/bazel/WORKSPACE:12:5: name 'http_archive' is not defined
ERROR: /usr/local/google/home/tcal/bazel/WORKSPACE:17:70: name 'git_hash' is not defined
ERROR: error loading package '': Encountered error while reading extension file 'dependency_support/dependency_support.bzl': no such package '@rules_hdl//dependency_support': error loading package 'external': Could not load //external package
```
So I replace `git_hash` with `rules_hdl_git_hash`, and added these two lines (which I found in another WORKSPACE file):
```
load("@bazel_tools//tools/build_defs/repo:http.bzl", "http_archive")
load("@bazel_tools//tools/build_defs/repo:utils.bzl", "maybe")
```

I then encountered:
```
ERROR: Failed to load Starlark extension '@rules_python//python:pip.bzl'.
Cycle in the workspace file detected. This indicates that a repository is used prior to being defined.
The following chain of repository dependencies lead to the missing definition.
 - @rules_python
This could either mean you have to add the '@rules_python' repository with a statement like `http_archive` in your WORKSPACE file (note that transitive dependencies are not added automatically), or move an existing definition earlier in your WORKSPACE file.
ERROR: cycles detected during target parsing
```

tcal-x, 2022-12-15:
I solved the last issue by pasting
```
load("@bazel_tools//tools/build_defs/repo:http.bzl", "http_archive")
http_archive(
    name = "rules_python",
    sha256 = "497ca47374f48c8b067d786b512ac10a276211810f4a580178ee9b9ad139323a",
    strip_prefix = "rules_python-0.16.1",
    url = "https://github.com/bazelbuild/rules_python/archive/refs/tags/0.16.1.tar.gz",
)
``` 
into my WORKSPACE, from https://github.com/bazelbuild/rules_python/releases/tag/0.16.1

However I continue running into issues; the next was needing to load `@rules_pkg`.   After addressing that, currently I'm missing `@rules_hdl_pip_deps_dataclasses_json`.   I'll stop here since I may be barking up the wrong tree.

mithro, 2022-12-15:
@QuantamHD / @hzeller - Any idea what is going wrong for @tcal-x ?

filmil, 2023-02-16:
Indeed, it would be nice to give a link to a repo that contains a minimal working example.

I created https://github.com/filmil/bazel-verilog-test/, and trying to fix it is just hitting one error after another.

I would expect that an example works out of the box.

filmil, 2023-02-16:
I got to here:
https://github.com/filmil/bazel-verilog-test/blob/main/WORKSPACE

Current status:

```
┬─[fmil@fmil9:~/code/test-verilog]─[02:47:15 AM]
│ (g/b:main)
╰─>$ bazel build //:counter_place_and_route
ERROR: /home/fmil/.cache/bazel/_bazel_fmil/e3baf8879015cdf440d7f2e54aae692b/external/bazel_tools/platforms/BUILD:89:6: in alias rule @bazel_tools//platforms:windows: Constraints from @bazel_tools//platforms have been removed. Please use constraints from @platforms repository embedded in Bazel, or preferably declare dependency on https://github.com/bazelbuild/platforms. See https://github.com/bazelbuild/bazel/issues/8622 for details.
ERROR: /home/fmil/.cache/bazel/_bazel_fmil/e3baf8879015cdf440d7f2e54aae692b/external/bazel_tools/platforms/BUILD:89:6: Analysis of target '@bazel_tools//platforms:windows' failed
ERROR: /home/fmil/.cache/bazel/_bazel_fmil/e3baf8879015cdf440d7f2e54aae692b/external/boost/BUILD.bazel:1847:14: errors encountered resolving select() keys for @boost//:stacktrace
ERROR: Analysis of target '//:counter_place_and_route' failed; build aborted: 
INFO: Elapsed time: 0.057s
INFO: 0 processes.
FAILED: Build did NOT complete successfully (0 packages loaded, 0 targets configured)
```

filmil, 2023-02-16:
This patch to bazel_rules_hdl gets me a bit further:

https://github.com/filmil/bazel_rules_hdl/commit/d55934ffaabec979404841d6fdeb6a83522bf524

```
╰─>$ bazel build //:counter_place_and_route                                           
INFO: Analyzed target //:counter_place_and_route (0 packages loaded, 0 targets configured).
INFO: Found 1 target...                    
ERROR: /home/fmil/.cache/bazel/_bazel_fmil/e3baf8879015cdf440d7f2e54aae692b/external/org_theopenroadproject/BUILD.bazel:1117:11: Compiling src/odb/src/db/dbObstruction.cpp [
for tool] failed: (Exit 1): gcc failed: error executing command (from target @org_theopenroadproject//:opendb_lib) /usr/bin/gcc -U_FORTIFY_SOURCE -fstack-protector -Wall -Wu
nused-but-set-parameter -Wno-free-nonheap-object -fno-omit-frame-pointer -g0 -O2 '-D_FORTIFY_SOURCE=1' -DNDEBUG -ffunction-sections ... (remaining 168 arguments skipped)

Use --sandbox_debug to see verbose messages from the sandbox and retain the sandbox build root for debugging
In file included from external/org_theopenroadproject/src/odb/include/odb/dbId.h:35,
                 from external/org_theopenroadproject/src/odb/include/odb/dbDiff.h:38, 
                 from external/org_theopenroadproject/src/odb/src/db/dbAttrTable.h:36, 
                 from external/org_theopenroadproject/src/odb/src/db/dbCore.h:47,
                 from external/org_theopenroadproject/src/odb/src/db/dbObstruction.h:35,
                 from external/org_theopenroadproject/src/odb/src/db/dbObstruction.cpp:33:
external/org_theopenroadproject/src/odb/include/odb/dbStream.h: In member function 'constexpr odb::dbOStream& odb::dbOStream::operator<<(const std::tuple<_Args2 ...>&) const
':
external/org_theopenroadproject/src/odb/include/odb/dbStream.h:197:14: error: binding reference of type 'odb::dbOStream&' to 'const odb::dbOStream' discards qualifiers
  197 |       return *this;
      |              ^~~~~
external/org_theopenroadproject/src/odb/include/odb/dbStream.h: In member function 'constexpr odb::dbIStream& odb::dbIStream::operator>>(std::tuple<_Args2 ...>&) const':
external/org_theopenroadproject/src/odb/include/odb/dbStream.h:423:14: error: binding reference of type 'odb::dbIStream&' to 'const odb::dbIStream' discards qualifiers
  423 |       return *this;
      |              ^~~~~
ERROR: /home/fmil/.cache/bazel/_bazel_fmil/e3baf8879015cdf440d7f2e54aae692b/external/org_theopenroadproject/BUILD.bazel:1117:11: Compiling src/odb/src/db/dbMetalWidthViaMap.
cpp [for tool] failed: (Exit 1): gcc failed: error executing command (from target @org_theopenroadproject//:opendb_lib) /usr/bin/gcc -U_FORTIFY_SOURCE -fstack-protector -Wal
l -Wunused-but-set-parameter -Wno-free-nonheap-object -fno-omit-frame-pointer -g0 -O2 '-D_FORTIFY_SOURCE=1' -DNDEBUG -ffunction-sections ... (remaining 168 arguments skipped
)

Use --sandbox_debug to see verbose messages from the sandbox and retain the sandbox build root for debugging
In file included from external/org_theopenroadproject/src/odb/include/odb/dbId.h:35,
                 from external/org_theopenroadproject/src/odb/include/odb/dbDiff.h:38, 
                 from external/org_theopenroadproject/src/odb/src/db/dbAttrTable.h:36, 
                 from external/org_theopenroadproject/src/odb/src/db/dbCore.h:47,
                 from external/org_theopenroadproject/src/odb/src/db/dbMetalWidthViaMap.h:36,
                 from external/org_theopenroadproject/src/odb/src/db/dbMetalWidthViaMap.cpp:34:
external/org_theopenroadproject/src/odb/include/odb/dbStream.h: In member function 'constexpr odb::dbOStream& odb::dbOStream::operator<<(const std::tuple<_Args2 ...>&) const
':
external/org_theopenroadproject/src/odb/include/odb/dbStream.h:197:14: error: binding reference of type 'odb::dbOStream&' to 'const odb::dbOStream' discards qualifiers
  197 |       return *this;
      |              ^~~~~
external/org_theopenroadproject/src/odb/include/odb/dbStream.h: In member function 'constexpr odb::dbIStream& odb::dbIStream::operator>>(std::tuple<_Args2 ...>&) const':
external/org_theopenroadproject/src/odb/include/odb/dbStream.h:423:14: error: binding reference of type 'odb::dbIStream&' to 'const odb::dbIStream' discards qualifiers
  423 |       return *this;
      |              ^~~~~
Target //:counter_place_and_route failed to build
Use --verbose_failures to see the command lines of failed build steps.
INFO: Elapsed time: 4.258s, Critical Path: 3.79s
```

Gotta stop for the day.

filmil, 2023-02-16:
Removed lots of errors (but not all) with this change:
https://github.com/filmil/bazel-verilog-test/commit/ae76240c176bf8057c28123ae4535f8200aeafac

filmil, 2023-02-17:
Still more issues (but again not all) removed by copying over the parts of `.bazelrc` that have to do with local C++ toolchain used.

https://github.com/filmil/bazel-verilog-test/commit/ae76240c176bf8057c28123ae4535f8200aeafac

filmil, 2023-02-17:
OK, after some running around, I have a working version of the example from the README.md.

See here: https://github.com/filmil/bazel-verilog-test

It's a bit more involved than necessary to set everything up. 
Read the [README.md][rm] for the precise instructions, 
they aren't overly complex.

[rm]: https://github.com/filmil/bazel-verilog-test/blob/main/README.md