hdl/bazel_rules_hdl #126: Build error; windows-related change breaking linux?

tcal-x, 2022-12-20:
On tip of current main, in root of the repo:

```
$ bazel build synthesis/tests:verilog_counter_synth 
Starting local Bazel server and connecting to it...
ERROR: /usr/local/google/home/tcal/.cache/bazel/_bazel_tcal/ee664ba5f06356c0566bacadf9a7ef5f/external/bazel_tools/platforms/BUILD:89:6: in alias rule @bazel_tools//platforms:windows: Constraints from @bazel_tools//platforms have been removed. Please use constraints from @platforms repository embedded in Bazel, or preferably declare dependency on https://github.com/bazelbuild/platforms. See https://github.com/bazelbuild/bazel/issues/8622 for details.
ERROR: /usr/local/google/home/tcal/.cache/bazel/_bazel_tcal/ee664ba5f06356c0566bacadf9a7ef5f/external/bazel_tools/platforms/BUILD:89:6: Analysis of target '@bazel_tools//platforms:windows' failed
ERROR: /usr/local/google/home/tcal/.cache/bazel/_bazel_tcal/ee664ba5f06356c0566bacadf9a7ef5f/external/at_clifford_yosys/BUILD.bazel:217:10: While resolving toolchains for target @at_clifford_yosys//:yosys: invalid registered toolchain '@bazel_skylib//toolchains/unittest:cmd_toolchain': 
ERROR: Analysis of target '//synthesis/tests:verilog_counter_synth' failed; build aborted: 
INFO: Elapsed time: 16.644s
INFO: 0 processes.
FAILED: Build did NOT complete successfully (39 packages loaded, 570 targets configured)
    Fetching repository @com_google_skywater_pdk; starting
    Fetching ...cal/google/home/tcal/.cache/bazel/_bazel_tcal/ee664ba5f06356c0566bacadf9a7ef5f/external/com_google_skywater_pdk; Extracting 3d7617a1acb92ea883539bcf22a632d6361a5de4.tar.gz
```

hzeller, 2022-12-20:
There were recent changes with Bazel 6.0.0 that seem to break things.
If you can, maye use bazel 5.x for now.

tcal-x, 2022-12-20:
> There were recent changes with Bazel 6.0.0 that seem to break things. If you can, maye use bazel 5.x for now

I was in fact using version 6.0.0.   I'll try installing 5.x.   Thanks!

closed 2022-12-20 by tcal-x