hdl/bazel_rules_hdl #128: pcre failing certificate?

renau, 2022-12-28:
I get this error when trying to build:

```
ERROR: An error occurred during the fetch of repository 'org_pcre_ftp':
   Traceback (most recent call last):
        File "XXX/_bazel_renau/3f7dad3b9f25e1dda25385629e7e4b4b/external/bazel_tools/tools/build_defs/repo/http.bzl", line 125, column 45, in _http_archive_impl
                download_info = ctx.download_and_extract(
Error in download_and_extract: java.io.IOException: Error downloading [https://ftp.exim.org/pub/pcre/pcre-8.44.tar.gz] to XXX/_bazel_renau/3f7dad3b9f25e1dda25385629e7e4b4b/external/org_pcre_ftp/temp7711447042372617183/pcre-8.44.tar.gz: No subject alternative DNS name matching ftp.exim.org found.
```

If I try to get it directly:
```
 wget https://ftp.exim.org/pub/pcre/pcre-8.44.tar.gz
--2022-12-27 17:43:03--  https://ftp.exim.org/pub/pcre/pcre-8.44.tar.gz
Resolving ftp.exim.org (ftp.exim.org)... 37.120.190.30, 2a03:4000:6:b381::2
Connecting to ftp.exim.org (ftp.exim.org)|37.120.190.30|:443... connected.
The certificate's owner does not match hostname ‘ftp.exim.org’
```

Any suggestion?

QuantamHD, 2022-12-28:
Hmm, it's possible that that particular mirror's certificate expired.

What operating system and version are you on. Sometimes the root certs on particular Linux distros expire and need to be updated

QuantamHD, 2022-12-28:
I'm away from my computer, but if you can find another link to pcre 8.44 you can add it here and it should resolve the issue

https://github.com/hdl/bazel_rules_hdl/blob/b3268b8c22c3da4bb577b0bfa5275fd508aab601/dependency_support/org_pcre_ftp/org_pcre_ftp.bzl#L25