hdl/bazel_rules_hdl #137: Compare and understand interfacing with OpenTitan

mithro, 2023-01-10:
OpenTitan has apparently adopted bazel in some way? See how they are using it and what we can do it?

hcindyl, 2023-08-16:
I spent some time trying to use `@verilator//:verilator_executable` from this project to be integrated with Opentitan to build https://github.com/lowRISC/opentitan/blob/0c2cd23c93338317c3fcb684b52c6e0ed256e659/hw/BUILD#L36

And here are some of the problems
1) Need to add the genrule to copy the wrapper `verilator` to the output `bazel-bin/external/verilator` directory
2) `verilator_executable` has to be called `verilator_bin` to be used in the wrapper (otherwise an environment variable needs to be set, which is a different one from what's defined by edalize)
3) The version prefix `v` in https://github.com/hdl/bazel_rules_hdl/blob/18bb869bd72af967aecd665ce215db308e5a69ec/dependency_support/verilator/verilator.bzl#L31-L37 actually breaks OT's verilator version check in https://github.com/lowRISC/opentitan/blob/0c2cd23c93338317c3fcb684b52c6e0ed256e659/util/check_tool_requirements.py#L142
4) OT uses GCC as the host compiler, when fall back to verilator 4.210 (the OT recommended version) we need some extra patching on src/V3Const.cpp (`#include <memory>`).
5) OT assumes `verilator` is in the system PATH, so there are extra patching works required for bazel to find the executables `verilator` and  `verilator_bin`
6) Even with all the aforementioned patches, the target is still failed to build with the following errors (in both verilator `4.210` and `4.224`)
```
INFO: verilator -f lowrisc_dv_chip_verilator_sim_0.1.vc -DDISABLE_PRIM_CDC_RAND_DELAY --trace --trace-fst --trace-structs --trace-params --trace-max-array 1024 --unroll-count 512 -CFLAGS " -std=c++11 -Wall -DVM_TRACE_FMT_FST -DVL_USER_STOP -DTOPLEVEL_NAME=chip_sim_tb" -LDFLAGS " -pthread -lutil -lelf" -Wall --threads 4 -Wno-fatal --threads 4
make -j 4 -f Vchip_sim_tb.mk
make[1]: Entering directory '.../execroot/lowrisc_opentitan/bazel-out/k8-fastbuild/bin/hw/build.verilator_real/sim-verilator'
make[1]: Leaving directory '.../execroot/lowrisc_opentitan/bazel-out/k8-fastbuild/bin/hw/build.verilator_real/sim-verilator'

ERROR: %Warning-ASSIGNDLY: ../src/lowrisc_systems_ast_0.1/rtl/vcc_pgd.sv:39:33: Unsupported: Ignoring delay on this assignment/primitive.
   39 |     vcc_pok_o <= #(ast_bhv_pkg::VCC_POK_RDLY) gen_supp_a;
      |                                 ^~~~~~~~~~~~
                    ... For warning description see https://verilator.org/warn/ASSIGNDLY?v=0.000
                    ... Use "/* verilator lint_off ASSIGNDLY */" and lint_on around source to disable this message.
%Warning-ASSIGNDLY: ../src/lowrisc_systems_ast_0.1/rtl/vcc_pgd.sv:42:33: Unsupported: Ignoring delay on this assignment/primitive.
   42 |     vcc_pok_o <= #(ast_bhv_pkg::VCC_POK_FDLY) gen_supp_a;
      |                                 ^~~~~~~~~~~~
%Warning-ASSIGNDLY: ../src/lowrisc_systems_ast_0.1/rtl/vio_pgd.sv:39:33: Unsupported: Ignoring delay on this assignment/primitive.
   39 |     vio_pok_o <= #(ast_bhv_pkg::VIO_POK_RDLY) gen_supp_a;
      |                                 ^~~~~~~~~~~~
%Warning-ASSIGNDLY: ../src/lowrisc_systems_ast_0.1/rtl/vio_pgd.sv:42:33: Unsupported: Ignoring delay on this assignment/primitive.
   42 |     vio_pok_o <= #(ast_bhv_pkg::VIO_POK_FDLY) gen_supp_a;
      |                                 ^~~~~~~~~~~~
%Warning-ASSIGNDLY: ../src/lowrisc_systems_ast_0.1/rtl/vcaon_pgd.sv:38:35: Unsupported: Ignoring delay on this assignment/primitive.
   38 |     vcaon_pok_o <= #(ast_bhv_pkg::VCAON_POK_RDLY) gen_supp_a;
      |                                   ^~~~~~~~~~~~~~
```
Likely related to #16

mithro, 2023-08-18:
FYI - Those version of Verilator are pretty old -- Verilator 4.210 2021-07-07 -- https://verilator.org/guide/latest/changes.html

hcindyl, 2023-08-18:
True.  The ToT of this project uses v.224

https://github.com/hdl/bazel_rules_hdl/blob/main/dependency_support/verilator/verilator.bzl#L59

which has the same build error.