hdl/bazel_rules_hdl #140: Build failure on Mac OS

rachitnigam, 2023-02-01:
I'm trying to build [XLS](https://github.com/google/xls) on Mac OS from source which pulls rules_hdl as a dependency and the build seems to fail with the following error:
```
ERROR: /private/var/tmp/b3261ecaf5b6ec09d8051d16006b13a8/external/com_icarus_iverilog/BUILD.bazel:351:10: Linking external/com_icarus_iverilog/ivl failed: (Exit 1): cc_wrapper.sh failed: error executing command (from target @com_icarus_iverilog//:ivl) external/llvm_toolchain/bin/cc_wrapper.sh @bazel-out/darwin-opt/bin/external/com_icarus_iverilog/ivl-2.params
```

Digging into it, it seems that `ld` on mac has the option `-no_pie` instead of `-no-pie` used here:
https://github.com/hdl/bazel_rules_hdl/blob/main/dependency_support/com_icarus_iverilog/bundled.BUILD.bazel#L375

I'm not super familiar with bazel but if there is an obvious fix, I can try to write up a PR

QuantamHD, 2023-02-01:
This is actually a rather easy fix if you want to give it a try.

The select below already has options for darwin(MacOS) in a list. You would just need to move that flag into the list. The select chooses to return a the list based on which platform your on.

```
select({
        "@bazel_tools//src/conditions:darwin": ["-Wl,-export_dynamic", "-no_pie"],
        "//conditions:default": ["-Wl,--export-dynamic", "-no-pie"],
    }),
```

rachitnigam, 2023-02-01:
Sweet! On it. Is there no need for a `"-Wl,-no_pie"` inside the select?

QuantamHD, 2023-02-01:
I suspect "-Wl,-no_pie" is required. I just didn't type it correctly

rachitnigam, 2023-02-01:
Updated the PR to use it!

PR #141 (2023-02-01): Use `-no_pie` for darwin builds
Fixes #140
closed 2023-02-01 by QuantamHD