hdl/bazel_rules_hdl #168: please add pczarnecki@ as a colaborator

proppy, 2023-06-07:
@QuantamHD is that possible to pczarnecki@ as a collaborator?

Context: as part of https://github.com/google/xls/issues/996 they'll need to be able to trigger the cloud build job thsembles, in order iterate quickly on their PRs.

/cc @tmichalak

tmichalak, 2023-06-07:
The handle is @lpawelcz , but this would be much appreciated.

umarcor, 2023-06-07:
I created teams bazel-triagers and bazel-contributors in https://github.com/orgs/hdl/teams and I invited the people that currently has triage or write access through direct invitation to this repo. I gave those teams the corresponding access in this repo. I invited @lpawelcz to team bazel-contributors (I believe he has write access already/now).

lpawelcz, 2023-06-07:
Thanks @proppy, @umarcor!