hdl/bazel_rules_hdl #172: Add tests for ASIC flow with ASAP7

lpawelcz, 2023-07-19:
From what I see, when it comes to ASAP7 support in `bazel_rules_hdl` the tests cover only the synthesis step:
https://github.com/hdl/bazel_rules_hdl/blob/e30c65c618dabffdcdeb45ad66e27fbaa2dc573e/synthesis/tests/BUILD#L71-L78

I would be good to add tests that will run `place_and_route` and `gds_write` flows with the results from `verilog_counter_asap7_synth` as input.

I started work on adding this but I encountered errors in Clock Tree Synthesis, here is the log:
```
[INFO ORD-0030] Using 12 thread(s).
[INFO CTS-0049] Characterization buffer is: BUFx4_ASAP7_75t_R.
[INFO CTS-0039] Number of created patterns = 11880.
[INFO CTS-0084] Compiling LUT.
Min. len    Max. len    Min. cap    Max. cap    Min. slew   Max. slew
2           8           1           34          1           64          
[WARNING CTS-0043] 1584 wires are pure wire and no slew degradation.
TritonCTS forced slew degradation on these wires.
[INFO CTS-0046]     Number of wire segments: 7500.
[INFO CTS-0047]     Number of keys in characterization LUT: 1744.
[INFO CTS-0048]     Actual min input cap: 1.
[WARNING CTS-0083] No clock nets have been found.
[INFO CTS-0008] TritonCTS found 0 clock nets.
[WARNING CTS-0082] No valid clock nets in the design.
[INFO RSZ-0058] Using max wire length 294um.
Placement Analysis
---------------------------------
total displacement          0.0 u
average displacement        0.0 u
max displacement            0.0 u
original HPWL             240.9 u
legalized HPWL            245.8 u
delta HPWL                    2 %

[INFO RSZ-0033] No hold violations found.
No paths found.
Placement Analysis
---------------------------------
total displacement          0.0 u
average displacement        0.0 u
max displacement            0.0 u
original HPWL             245.8 u
legalized HPWL            245.8 u
delta HPWL                    0 %

[INFO DPL-0001] Placed 253 filler instances.
[WARNING DPL-0005] Overlap check failed (248).
[ERROR DPL-0033] detailed placement checks failed.
```

CC @proppy

lpawelcz, 2023-07-20:
It is possible to finish the flow by moving the `check_placement` step before `filler_placement` so that the check won't fail. However the result GDS looks a lot different than one I got straight from ORFS flow. PDN looks quite different and the filler cells are placed only along the power rails and are limited to only one cell type: `FILLERxp5_ASAP7_75t_R` while in ORFS flow there are more cell types and are placed uniformly throughout the whole die: `FILLERxp5_ASAP7_75t_R`, `FILLER_ASAP7_75t_R`, `DECAPx1_ASAP7_75t_R`, `DECAPx2_ASAP7_75t_R`, `DECAPx4_ASAP7_75t_R`, `DECAPx6_ASAP7_75t_R`, `DECAPx10_ASAP7_75t_R`.

I noticed that ASAP7 version used in `bazel_rules_hdl` is very old ([The-OpenROAD-Project/asap7/157c92cf](https://github.com/The-OpenROAD-Project/asap7/tree/157c92cfd2567a98c5f794fb10c2dd0713516374)) which could be the cause of the differences between layouts. 
In various places in PDK and bazel_rules for ASAP7 there are also scaling operations (multiply by 4) - I'm not sure what is their purpose but it is also possible that causes overlapping after `filler_placement`. I will try to bump the PDK and investigate this scaling.

proppy, 2023-07-24:
> ASAP7 comes with 2 variants of the technology: regular and scaled (4 times bigger). However, ASAP7 does not provide all required files for the regular variant (e.g. technology LEF), all files required for RTL->GDS flow are available only for the scaled variant.
> ORFS uses regular variant. It doesn't have ASAP7 pinned as submodule or anything like that. Instead of that, at some point in history, the most important files from this PDK were imported straight into ORFS repository. Missing files for the regular variant were imported from scaled variant and manually converted to match regular variant. Then those files were updated with changes from upstream ASAP7 repository and probably with some additional external changes. Because of that it is hard to tell which version of ASAP7 is the closest to one used in ORFS.

Is there already an issue in the ASAP7 repo to discuss reconciling the two version? It sounds like we should be able to import ORFS version in the ASAP7 repo since it's incomplete there.

> I will attach here GDS files for comparison.

I think it would also be useful to compare the TCLs generated by bazel_rules_hdl with the ones coming from ORFS for ASAP7, to see how the two flows differ.

lpawelcz, 2023-07-24:
> Is there already an issue in the ASAP7 repo to discuss reconciling the two version? It sounds like we should be able to import ORFS version in the ASAP7 repo since it's incomplete there.

There was https://github.com/The-OpenROAD-Project/asap7/issues/19 which touched the problem. There is a [comment](https://github.com/The-OpenROAD-Project/asap7/issues/19#issuecomment-1115712485) there that states the files are missing due to licensing issues.

> I think it would also be useful to compare the TCLs generated by bazel_rules_hdl with the ones coming from ORFS for ASAP7, to see how the two flows differ.

I will check  TCL scripts, compare them and post the results.

proppy, 2023-07-25:
@QuantamHD for visilibility

proppy, 2023-07-26:
> flow still fails on `placement_check`

do you mean `check_placement`?

proppy, 2023-07-26:
> for some reason the placement check wont fail in case of ORFS flow.

did you already compare the `*_place_and_route_commands.tcl` file generated in `bazel-bin` with the ORFS TCL scripts in https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/tree/master/flow/scripts?

lpawelcz, 2023-07-26:
> do you mean check_placement?

Yes, sorry

> did you already compare the *_place_and_route_commands.tcl file generated in bazel-bin with the ORFS TCL scripts in https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/tree/master/flow/scripts?

In bazel_rules_hdl filler cell placement is done in the same script as clock tree synthesis.
`clock_tree_synthesis` command has additional `-post_cts_disable` arg which seems to be obsolete, while in ORFS there are additional args: `-sink_clustering_size`, `-sink_clustering_max_diameter`, `-distance_between_buffers` and `-balance_levels`
Then there is `set_placement_padding` which had different values in bazel flow (`-left 2 -right 2`) and ORFS (`-left 1 -right 1`), I tried to set values from ORFS but it didn't change anything.

I will try to use ASAP7 files from ORFS and check this `gdstxt` feature of klayout

lpawelcz, 2023-07-26:
@proppy I was able to finish the flow and pass  `check_placement` with setup described in https://github.com/hdl/bazel_rules_hdl/issues/172#issuecomment-1651225432 but without specifying more filler cell types in ASAP7 open_road_pdk_configuration (there was only `FILLERxp5_ASAP7_75t_R`).

Now I will focus on isolating the fix which is probably one of the changes described in the linked comment. Then I will try to answer the question: Why those additional filler cells cause placement check failures in bazel flow?
For that I will compare filler cells definitions from ORFS files with definitions from ASAP7 repository.

lpawelcz, 2023-08-24:
In order to enable processing designs targeting ASAP7 PDK and to achieve results similar to `OpenROAD-flow-scripts` it is required to:
* use non-scaled variant of the PDK,
* extend the list of the spacer cells available for filler placement stage.

With recent updates to ASAP7:
* https://github.com/The-OpenROAD-Project/asap7/pull/54
* https://github.com/The-OpenROAD-Project/asap7/pull/57
we are able to use the non-scaled variant of the PDK which is used by default in ORFS.

closed 2023-09-07 by QuantamHD