hdl/bazel_rules_hdl #207: Feature request: create a small example project for OpenROAD asap7

oharboe, 2023-10-12:
It would be great to have an example project on how to use bazel_rules_hdl for OpenROAD and asap7.

mithro, 2023-10-16:
I agree.

proppy, 2023-10-18:
I think @lpawelcz recently did something for https://github.com/google/xls/pull/1031, it should be possible to generalize as a more generic example.

lpawelcz, 2023-10-18:
Hi, there are some examples in `//synthesis/tests`. Please take a look at [BUILD](https://github.com/hdl/bazel_rules_hdl/blob/main/synthesis/tests/BUILD) file, there are 2 simple examples of physical design flow which uses OpenROAD.
One targets SKY130, the other ASAP7. For example `bazel build //synthesis/tests:counter_asap7_asic` will run physical design flow targeting ASAP7 on counter design.
When it comes to the configuration, PDK is specified in [synthesize_rtl](https://github.com/hdl/bazel_rules_hdl/blob/7fd311eacc3352831cbae8b83dc0ed02b45e14b6/synthesis/tests/BUILD#L84-L92) rule with `standard_cells` attribute.

mithro, 2023-10-26:
There are;
```
//synthesis/tests:counter_asap7_asic
//synthesis/tests:counter_asap7_place_and_route
//synthesis/tests:counter_asic
//synthesis/tests:counter_place_and_route
//synthesis/tests:verilog_adder
//synthesis/tests:verilog_adder_synthesized
//synthesis/tests:verilog_counter
//synthesis/tests:verilog_counter_asap7_synth
//synthesis/tests:verilog_counter_asap7_synth_sta
//synthesis/tests:verilog_counter_synth
//synthesis/tests:verilog_counter_synth_sta
```
and
```
//flows/tests:synth_sky130_adder
//flows/tests:synth_sky130_adder_with_clock_period
//flows/tests:synth_sky130_smoke
//flows/tests:synth_sky130_smoke_0__deps
//flows/tests:synth_sky130_smoke_1__deps
//flows/yosys:synth_sky130
//flows/yosys:synth_sky130_bin
//flows/yosys:synth_sky130_pkg
```

oharboe, 2023-10-26:
> There are;
>
> ```
> //synthesis/tests:counter_asap7_asic
> //synthesis/tests:counter_asap7_place_and_route
> //synthesis/tests:counter_asic
> //synthesis/tests:counter_place_and_route [...]

Does any of these create a macro that is used by a subsequent design?

QuantamHD, 2023-10-26:
@oharboe That's not something we've built yet, but it wouldn't be too hard to accommodate.

oharboe, 2023-10-26:
> @oharboe That's not something we've built yet, but it wouldn't be too hard to accommodate. Do you have a motivating use case?

👍 To support such macro artifacts was my motivation for creating a wafer thin Bazel layer on top of ORFS + I also need the ORFS `make issue` feature. 

We have also started to use bazel_rules_hdl to build Verilaror binaries.

oharboe, 2023-11-02:
Another use-case that I am trying to add to my wafer thin layer on top of Bazel is to be able to view DRC reports.

Is this something that is supported with bazel_rules_hdl?

I.e. when global routing fails, I want to start the OpenROAD GUI and load the congestion.rpt file in the DRC viewer.

```
[ERROR GRT-0119] Routing congestion too high. Check the congestion heatmap in the GUI and load bazel-out/k8-fastbuild/bin//build/reports/asap7/BoomCore/base/congestion.rpt in the DRC viewer.
Error: global_route.tcl, 27 GRT-0119
Command exited with non-zero status 1
```