hdl/bazel_rules_hdl #295: place_and_route: Syntax of generated textproto files

mczyz-antmicro, 2024-03-11:
Textproto files are generated from `generate_power_results` and `generate_area_results`, which are used in the [global_routing.tcl](https://github.com/hdl/bazel_rules_hdl/blob/main/place_and_route/private/global_routing.bzl#L75). What syntax are they meant to be? By file extension, I presume that they are meant to be [protobuf files](https://protobuf.dev/), but unfortunately they are not consistent with protobuf syntax (you can test with [protolinter](https://github.com/yoheimuta/protolint)). I propose to make effort to make these files consistent.

Example of generated file:
```
total_package_power_watts: 0.0059173996560275555
corner: "asap7-sc7p5t_rev28_rvt-ccs_ss_SS"
power_maginitude: "p"
```

QuantamHD, 2024-03-11:
They are text protos. https://protobuf.dev/reference/protobuf/textformat-spec/

closed 2024-03-11 by mczyz-antmicro

QuantamHD, 2024-03-11:
https://protobuf.dev/reference/cpp/api-docs/google.protobuf.text_format/

If you want to parse them