hdl/bazel_rules_hdl #336: Getting ready for bzlmod

hzeller, 2024-06-29:
It would be good to get bazel_ruloes_hdl ready for bzlmod and add a `MODULE.bazel`.

https://bazel.build/external/migration

The oldest currently still maintained bazel 5 (and that can only work with old-school WORKSPACE ) is [on the way out at the end of the year ](https://bazel.build/release#support-stages), bazel 6 allows to have bzlmod enabled and bazel 7 enables it by default.

MaxwellBo, 2024-09-01:
Are there any updates here? I'm finding it very painful to use this with modules and would appreciate some guidance on the correct approach to do so in the interim.

hovind, 2025-02-21:
I have done some experimenting over at https://github.com/hdl/bazel_rules_hdl/compare/main...hovind:bazel_rules_hdl:bzlmod/bazel-8, in case it is useful for anyone. `bazelisk test ...` passes, but it's far from "production quality".

hovind, 2025-02-21:
> Nice! I also started working on the first small steps towards it. But if you're working on it: do you plan to make a PR, [@hovind](https://github.com/hovind) ?

I don't have any immediate plans to make a PR. I've had most of this code laying around since May 2024 to give you an idea of the lack of progress, it's just something I've dabbled in every now and then in my spare time to help myself learn a bit of `bzlmod`. Just thought I'd share in case it were useful to anyone. :)

hzeller, 2025-02-21:
ok. I'll start chipping away on it then with little PRs here and there.

hovind, 2025-02-21:
> ok. I'll start chipping away on it then with little PRs here and there.

That sounds like a good approach, I'm looking forward to following your progress!

freeform-andre, 2025-03-01:
Would it make sense to shard the repo up into smaller repositories and publish those as bzlmod modules first? The repo has quite a collection of tools in it. Maybe it would be easier to allocate time to publish things one at a time and ensure all the shared intermediates are available before tackling the larger parts of the repo.

mgottscho, 2025-04-25:
While we wait for native bzlmod support, figured I'd share how we bridge the gap in the https://github.com/xlsynth/bedrock-rtl repo using an extension:

https://github.com/xlsynth/bedrock-rtl/blob/main/MODULE.bazel#L24-L25
https://github.com/xlsynth/bedrock-rtl/blob/main/dependency_support/rules_hdl/extension.bzl

Note that we are currently only using the `verilog_library` rule in bedrock-rtl; YMMV for doing other stuff with our approach.

j2kun, 2025-06-04:
For one, net_zlib is already on the BCR: https://registry.bazel.build/modules/zlib

MrAMS, 2025-11-17:
> [@MrAMS](https://github.com/MrAMS) Great! Thanks for trying this out! Indeed that file was present for pre-bzlmod dependency stuff. I wonder what is pulling that file again.

I don't know how to pinpoint who's pulling that file again. Do you have any suggestions?
If it's convenient for you, would you be able to provide a minimal working example project? Thank you very much.

MrAMS, 2025-11-17:
> Alright, the example seems to work now. Let me know if you are encountering any issues.

@lromor Thanks for the great demo project! Following your example, I successfully upgraded my project [bazel-chisel-demo](https://github.com/MrAMS/bazel-chisel-demo), and the [CI build passed](https://github.com/MrAMS/bazel-chisel-demo/actions/runs/19437232694/job/55611200641). I really appreciate your contribution and help!

While working on this, I identified a couple of related issues and opened two PRs: [lemon#1](https://github.com/amorphic-io/lemon/pull/1) and [yosys#1](https://github.com/amorphic-io/yosys/pull/1). Could you please take a look when you have a moment?

Separately, I'd like to start a discussion. I've noticed that `init_submodules = True` in the `git_override` seems to cause significant fetching times on each run. To optimize this, what are your thoughts on creating GitHub Releases and switching to `archive_override` instead?

lromor, 2025-11-17:
Great to hear! I'll have a look tomorrow. Init submodules was required because openroad had submodules, tomorrow I'll test if git archives will contain the submodules. The other issue is that git archive hashes do not seem to be always guaranteed to be unique.